PX4/PX4Devguide #12: Permission error when installing gitbook

CarlOlsson, 2015-11-02:
**When following the guide to install gitbook, running:**
npm install grunt-cli -g 

**gives**
npm http GET https://registry.npmjs.org/grunt-cli
npm http 304 https://registry.npmjs.org/grunt-cli
npm ERR! Error: EACCES, mkdir '/usr/local/lib/node_modules'
npm ERR!  { [Error: EACCES, mkdir '/usr/local/lib/node_modules']
npm ERR!   errno: 3,
npm ERR!   code: 'EACCES',
npm ERR!   path: '/usr/local/lib/node_modules',
npm ERR!   fstream_type: 'Directory',
npm ERR!   fstream_path: '/usr/local/lib/node_modules/grunt-cli',
npm ERR!   fstream_class: 'DirWriter',
npm ERR!   fstream_stack: 
npm ERR!    [ '/usr/lib/nodejs/fstream/lib/writer.js:171:23',
npm ERR!      '/usr/lib/nodejs/mkdirp/index.js:37:53',
npm ERR!      'Object.oncomplete (fs.js:107:15)' ] }
npm ERR! 
npm ERR! Please try running this command again as root/Administrator.

npm ERR! System Linux 3.19.0-31-generic
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "grunt-cli" "-g"
npm ERR! cwd /home/carl/Wingtra
npm ERR! node -v v0.10.25
npm ERR! npm -v 1.3.10
npm ERR! path /usr/local/lib/node_modules
npm ERR! fstream_path /usr/local/lib/node_modules/grunt-cli
npm ERR! fstream_type Directory
npm ERR! fstream_class DirWriter
npm ERR! code EACCES
npm ERR! errno 3
npm ERR! stack Error: EACCES, mkdir '/usr/local/lib/node_modules'
npm ERR! fstream_stack /usr/lib/nodejs/fstream/lib/writer.js:171:23
npm ERR! fstream_stack /usr/lib/nodejs/mkdirp/index.js:37:53
npm ERR! fstream_stack Object.oncomplete (fs.js:107:15)
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/carl/Wingtra/npm-debug.log
npm ERR! not ok code 0

LorenzMeier, 2015-11-02:
Depending on what the ubuntu defaults are you might need to use sudo. Please check the nom docs.

andre-nguyen, 2015-11-02:
I use Ubuntu and I can confirm that all you have to do is sudo before some of the commands.

CarlOlsson, 2015-11-02:
Yes it works with sudo. Just wanted to be sure since the devguide also states:
*Never ever fix permission problems by using 'sudo'. It will create more permission problems in the process and require a system reinstallation to fix them.*

When running the next command: **sudo npm install** I get:
npm ERR! install Couldn't read dependencies
npm ERR! Error: ENOENT, open '/home/carl/Wingtra/px4devguide/package.json'
npm ERR! If you need help, you may report this log at:
npm ERR!     <http://github.com/isaacs/npm/issues>
npm ERR! or email it to:
npm ERR!     <npm-@googlegroups.com>

npm ERR! System Linux 3.19.0-31-generic
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/carl/Wingtra/px4devguide
npm ERR! node -v v0.10.25
npm ERR! npm -v 1.3.10
npm ERR! path /home/carl/Wingtra/px4devguide/package.json
npm ERR! code ENOENT
npm ERR! errno 34
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/carl/Wingtra/px4devguide/npm-debug.log
npm ERR! not ok code 0

Will look into this tomorrow

LorenzMeier, 2015-11-04:
It sounds a bit like your local install is bad

CarlOlsson, 2015-11-06:
Now I think it works, creating a symbolic link solves the error:
sudo ln -fs /usr/bin/nodejs /usr/local/bin/node

closed 2016-01-13 by LorenzMeier