PX4/PX4Devguide #17: Permission to PX4/Devguide.git denied

CarlOlsson, 2015-11-11:
Tried to push some changes to the devguide and got this back:

remote: Permission to PX4/Devguide.git denied to CarlOlsson.
fatal: unable to access 'https://github.com/PX4/Devguide.git/': The requested URL returned error: 403

Have I done something wrong or do I need to get permission?

LorenzMeier, 2015-11-11:
You need to fork it and send a pull request, like usual.

CarlOlsson, 2015-11-11:
Yes I have done that

carl@carl-EliteBook:~/Wingtra/px4devguide/Devguide$ git status
On branch added_jlink_debugger
nothing to commit, working directory clean

carl@carl-EliteBook:~/Wingtra/px4devguide/Devguide$ git push origin added_jlink_debugger 
Username for 'https://github.com': CarlOlsson
Password for 'https://CarlOlsson@github.com': 
remote: Permission to PX4/Devguide.git denied to CarlOlsson.
fatal: unable to access 'https://github.com/PX4/Devguide.git/': The requested URL returned error: 403

mhkabir, 2015-11-11:
You're trying to push to the original repo.

Set up your remotes to have origin as your fork, and upstream as PX4/Devguide.

CarlOlsson, 2015-11-11:
Oh sorry! Thanks

closed 2016-01-13 by LorenzMeier