PX4/PX4Devguide #20: Toolchain Installation Snapdragon Flight

devbharat, 2015-11-16:
In the instructions provided in the devguide 

    sudo apt-get install gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf -y
    tar xf Hexagon.LNX.7.2\ Installer-07210.1.tar
    chmod u+x Hexagon.LLVM_linux_installer_7.2.10.bin
    ./Hexagon.LLVM_linux_installer_7.2.10.bin

It isn't really clear where to get the tar file from.

boosfelm, 2015-11-16:
i have it, it is uploaded to our drive, technical, resources, snapdragon.
please ping me if anyone else needs the file.

devbharat, 2015-11-17:
Is it not in public domain ? If not we can add that as a comment in the devguide and close this.

boosfelm, 2015-11-17:
not public atm, but I don't know wether it is critical. @LorenzMeier shall we provide a dropbox link and close this here?

LorenzMeier, 2015-11-18:
Its not public, so a TODO would make sense there.

boosfelm, 2015-11-19:
@devbharat can you please add a TODO to the dev guide and point to yourself if someone needs the binary for now.

devbharat, 2015-11-19:
Yes

Pykercao, 2015-12-16:
Hi,boosfelm
I'm a Amateur player of PX4, the same problem I met when I build development environment on Ubuntu.Could you send Hexagon.LNX.7.2 to me?
Best Regards

boosfelm, 2015-12-16:
@devbharat can you do it? Thanks

mcharleb, 2015-12-16:
You can request the SDK from https://developer.qualcomm.com/software/hexagon-dsp-sdk/application

Pykercao, 2015-12-17:
@mcharleb That means I can get Hexagon.LNX.7.2 from https://developer.qualcomm.com/software/hexagon-dsp-sdk/application if I request?

Pykercao, 2015-12-17:
Does anyone have Hexagon.LNX.7.2 ? Who can send the file to me ? I'm in urgent need of the file.
Best Regards,
Pykercao

mcharleb, 2015-12-17:
@Pykercao It may take a day or so to get the tools after you fill out the form.

rgreid, 2016-01-22:
Once we have access to this page: https://developer.qualcomm.com/software/hexagon-dsp-sdk/tools where should we look to get access to 'Hexagon.LNX.7.2\ Installer-07210.1.tar' ?

I have already downloaded 'qualcomm_hexagon_sdk_2_0_eval.bin' however I'm getting strange errors building with Hexagon Tools v6.4.06

mcharleb, 2016-01-22:
Send me your email  (charlebm@gmail.com) and I'll have someone send you the Hexagon.LNX.7.2\ Installer-07210.1.tar file.

mcharleb, 2016-01-22:
@rgreid I sent a request for you and cc you on the email.

smnogar, 2016-02-08:
I am also in need of this file. I would very much appreciate it if someone could send me a link.

LorenzMeier, 2016-02-22:
The file is available from Intrinsyc. We're working with Qualcomm to get it into the normal SDK tools space. Issue to track this is https://github.com/PX4/Devguide/issues/97

closed 2016-02-22 by LorenzMeier

joaonatividade, 2016-03-24:
Hi guys! I'm still trying to find the "Hexagon.LNX.7.2 Installer-07210.1.tar" file. Can you help me with this? I need it in order to start programming with my PX4.

Thanks in advance,

João Natividade

LorenzMeier, 2016-03-24:
Please contact @mcharleb.

chenbinpeng, 2016-04-26:
I have the same problem with Pykercao, Could someone send Hexagon.LNX.7.2 to me? Thanks very much!(cbp@gezhuzi.com)

kartikmohta, 2016-05-12:
Just to let you know, the `Hexagon SDK v2.0 Add-On for HVX - Linux` download on https://developer.qualcomm.com/software/hexagon-dsp-sdk/tools contains the Hexagon Tools v7.2.10. Just run it as `sh qualcomm_hexagon_sdk_hvx_addon_2_0_eval.bin` and it starts up the installer for the Hexagon Tools.

mcharleb, 2016-05-12:
Thanks! I'll update the docs.