PX4/PX4Devguide #29: ROS dev guide issues

LorenzMeier, 2015-11-24:
@andre-nguyen few things need to be fixed:

topic/service name: please remove starting /: it prevents namespacing (roslaunch <group ns="">) Also maybe add somewhere note that mavros/ prefix may be different in some advanced configurations (because all topics are in private namespace ~ which expands as <node_name>/, node name may be changed in launch).
Before starting operation better to wait connection: ~state provide that information, also you can check that OFFBOARD is actually enabled.
Topic queue_size, 1 is ok, but in ros guides usually 10...
One side effect that current ROS Service API is syncronious, so some calls may block for some time (usually mavros timeouts 5-10 sec). Perhaps promises from ROS2 may help to write single-threaded controller nodes, but that is future…
// offtopic: @LorenzMeier as i seen, firmware interprets all mode set requests for main mode only, so i can't request AUTO.RTL, only AUTO.MISSION.
Maybe i should add sanity check on that kind of modes? And second question: should i add RATTITUDE or it is experimental? I only see it in QGC...

andre-nguyen, 2015-11-24:
Sure thing, just to summarize neatly:

[ ] Remove starting "/" to allow namespacing.
[ ] Add tip about the prefix being dependent on the developer's set up. 
[ ] Listen in on ~state to check that offboard is enabled and connection to FCU is up.
[ ] Increase topic queue size to 10

For the last point, am I supposed to spawn the service call into another thread?

vooon, 2015-11-24:
@andre-nguyen for that example i think nope, but that one thing that developer should keep in mind.
Another possible way is opposite: make talker thread for setpoint.

commit 90ff9b68 (2015-11-28): fix #29. cleaned topic names and add some tips to offboard control
closed 2015-11-28 by LorenzMeier