PX4/PX4Devguide #34: The make problem on ubuntu 15.10

nephen, 2015-12-15:
I got it by manual re installing gcc-arm-none-eabi 4.8 on ubuntu 15.10, now I can make firmware completely
cd ~
wget https://launchpadlibrarian.net/186124160/gcc-arm-none-eabi-4_8-2014q3-20140805-linux.tar.bz2
tar -jxf gcc-arm-none-eabi-4_8-2014q3-20140805-linux.tar.bz2
exportline="export PATH=$HOME/gcc-arm-none-eabi-4_8-2014q3/bin:\$PATH"
if grep -Fxq "$exportline" ~/.profile; then echo nothing to do ; else echo $exportline >> ~/.profile; fi
. ~/.profile

julianoes, 2016-04-21:
Thanks for raising this. On Ubuntu 15.10 arm-none-eabi 4.9.3 instead of 4.8.4 is installed, we need to consolidate this.

@LorenzMeier What do you thing?

julianoes, 2016-07-27:
Old, closing.

closed 2016-07-27 by julianoes