PX4/PX4Devguide #59: Add a search function

julianoes, 2016-01-13:
A start would be to just link the search button to a google search containing: ```site:dev.px4.io```

LorenzMeier, 2016-01-13:
Gitbook has it. Its just the 3DR template which is broken because of the local hackery. @kaitlynhova You must be running into the same issue with the 3DR dev guide - how did you resolve it?

kaitlynhova, 2016-01-24:
Hey @LorenzMeier ! Actually,  I think Tim (tim@3drobotics.com) would be more familiar with this. I'll check in with him to see how he did it. Here is his [github profile](https://github.com/tcr3dr)

julianoes, 2016-02-16:
@tcr3dr our search button is broken. Any advice how to get it back? Thanks.

julianoes, 2016-04-27:
Bump! Anyone?

julianoes, 2016-07-13:
Resolved with move to gitbook.

closed 2016-07-13 by julianoes