PX4/PX4Devguide #67: Reminder: Flight Mode

tumbili, 2016-01-22:
Stabilise for fixed wing does not appear in the flight mode description.

tubeme, 2016-03-10:
As well as Acro mode for fixed wing des not appear. Actually Manual in FW = Stabilized flight, while we have to switch to Acro in order to have real Manual Flight. 
Her is what it is now:

MANUAL
Fixed wing aircraft/ rovers / boats: The pilot's control inputs are passed directly to the output mixer.

Multirotors:

ACRO: The pilot's inputs are passed as roll, pitch, and yaw rate commands to the autopilot. This allows the multirotor to become completely inverted. Throttle is passed directly to the output mixer
RATTITUDE The pilot's inputs are passed as roll, pitch, and yaw rate commands to the autopilot if they are greater than the mode's threshold. If not the inputs are passed as roll and pitch angle commands and a yaw rate command. Throttle is passed directly to the output mixer.
ANGLE The pilot's inputs are passed as roll and pitch angle commands and a yaw rate command. Throttle is passed directly to the output mixer.

Which is not exactly what it is.

It is very perplex and allays overwhelming for the novice. So know my question is what is the default mode in Manual for the multicopters? If I don't program acro switch neither rat switch what is the mode i'm in? In FW I'm in Manual Stabilized flight if I done have Acro switch.

I think that the current system of naming the modes is very contradictory thus hard to understand. 

What is Assisted Flight? is it manual or auto mode?? In terms or user experience Assisted flight is Auto mode as well, because the computer automatically does things for me.

So when you have to understand the difference between Assisted mode and Auto mode you get confused because you perceive both as Auto modes.

There should be simpler system for the modes convention and naming and explanation. I don't say change in the functioning and features but how they are called explained and experienced. Ease of use = larger user base. You can make a very complex system work in two ways, simple way and complex way depending on the user needs and experience.

So here is a simpler way of doing it:

**Manual** (without submode switch, ie without Acro,Rat, Angle switch)

For all vehicles the default function should be Manual Stabalized. This way Manual = Stabalized and all sub-modes of Manual are not mandatory. 

**Assisted**

In Assisted the default should be POSCTL by default. I never ever needed ALTCTL never used it nor i needed it. So ALTCTL can be invoked by missing GPS or other pos data, or if I have a sub-mode switch in this mode posctl switch and I mandatory switch it to ALTCTL.

**Mission**

It is pretty explanatory. Just start mission execution.

AUTO_LOITER - should become just LOITER. It is my first line of defense, it is my yellow distress call. Especialy if I'm beginner. I use it if I have to pause and take a breath in a situation, and then continue on. It is actually a PUSE button. So It should work in all modes! even in the Acro modes. (You can fly acro and still have GPS lock or pos data.)

AUTO_RTL should become RTL. It is my distress red call so you cannot make it as a submode of Auto!!! Usually when i hit RTL I'm desperate... something is happening. It should work from every mode. Now it works from the assisted modes if GPS lock and from Auto mode and why is it then in Auto when it works from both Auto and Assisted?

That's it plain and simple. We have Mode switch with 3 modes Man / Ass / Miss, Manual switch, Assisted switch, Loiter Switch and RTL switch if I'm advanced.

If I'm novice I set Mode switch with 3 modes Man / Ass / Miss. And I can set up either RTL and Loiter switches or make a Emergency 3 pos switch with RTL in the middle position as a first line of defense and RTL in on position as a second line of defense.

This way as a novice I will have just two switches. And with this two switches with the defaults I gave above I will have pretty much all that I need from my drone being FW, Multi or VTOL. For the VTOL we have just Transition switch.

tubeme, 2016-03-10:
I'm thinking how to explain the modes in the users manual with icon graphics since last week. And I try find a way to explain it to a person that does not have any knowledge on drones. And believe me I cannot figure it out how to do it. I think the system will become monster with thousand of modes and sub modes and sub modes and yes for me that is very good but for the novice it is not.

The main success of DJI it not their machines but their simple 3 MODES for flight. 
Manual = Stabalized, GPS, RTH. You need nothing more in 80-90% of the situations.

Of course Loiter is mandatory for the Airplanes. But for Multicopters you don't even need Loiter because POSCTL is the same as loiter in terms of dynamics. The only difference is the gimbals does not work in Loiter vs POSCTL.

GervaisK, 2016-03-10:
> Links:

> The main success of DJI it not their machines but their simple 3 MODES 
> for flight. Manual = Stabalized, GPS, RTH. You need nothing more in 
> 80-90% of the situations.

No, DJI is using MANUAL (which is Stabilize in fact), ATTI (AltCtl), GPS 
since the GPS for the NAZA came out.Before that it was MANUAL and ATTI. 
Recent Phantom 4 is coming with P (Position),S (Sport), both GPS modes, 
A (ATTI)

Althold , or AltCtl. is mandatory as a training/trial Flight Mode esp. 
for new builds or beginners, which usually know little and tend to start 
with GPS and RTL, which often turns out into a desaster, due to TBE or 
bad GPS reception. Relying a security which wasn´t there. IRIS with RTL 
on 3rd position of FM switch was a joke, which caused some crashes.

If smooth camera flights w/o GPS correction movements are required, 
Althold is mandatory as well.

So if DJI sucess matters (not a bad idea) one could leave Stabilize 
away.Never ever Althold. Btw. lately the Phantom 4 came with a Pause 
Buttom (they copied of the 3DR SOLO, which had a lot of probs with it in 
the beginning because users did not realize,that this is depending on 
GPS reception) The idea is sort of Panic button.

Am 2016-03-10 01:48, schrieb tubeme:

tubeme, 2016-03-10:
@GervaisK 

I moved the issue here https://github.com/PX4/Firmware/issues/3967

Yes you are correct. Probably with that modes I got got introduced to their controllers. So most people I know use GPS, Atti, RTH. Most people that fly manual i know don't use DJI. And yes I agree AltCtl is mandatory for closed environments where there is no GPS signal.

DJI changed their GPS name convention, because they implement more ways of pos control with different sensors, so it is inadequate to keep the name GPS mode. 

My point though was not what exactly their modes are but rather that they have again 3 pos switch concept. + the addition of Emergency switch. So now they have 3pos  P / S / A of which P and S pos modes and A AltCtl. In your right hand you have the emergency switch. Simple to learn, simple to explain and you get very very fast the unconscious competence of reacting in all situations. 

DJI success in my opinion maters, because the rule is If you want a technology to become useful for the people it should be affordable (in terms of price and in terms of easy adoption.). DJI did it by the book, technologically wise and marketing and sales wise. The results are ~1 billion USD turnover for 2015. It makes more drones than all the other players together. It does not matter they dont produce the best drones. They produce upper medium level machines. But reliable and easy to use. Their concept is seen in the other products they make like the Ronin and Zenmuse gimbals. It is not the best gimbal but is the most easiest to balance, thus the fastest for work.

Analyzing success is equally important as analyzing failure, but in reality is very neglected.

I love Pixhawk and PX4, hate DJI but I have to agree that they are a bench mark in the industry. So we have to follow on their good ideas the same way they are following on our good ideas.