PX4/PX4Devguide #68: Mavros FCU connection problem in Gazebo

chutsu, 2016-01-27:
tdlr: The docs need updating!

Having followed your docs simulating the firmware using Gazebo (link [here](http://dev.px4.io/simulation-ros-interface.html)), I and others (mavros issue [here](https://github.com/mavlink/mavros/issues/476)) have failed to connect `mavros` to obtain data from ros topics (e.g. /mavros/altitude etc..).

The reason for this as github user `nachi1992` has pointed out, the UDP receive port in the docs is wrong. Instead of port `14540` it should be **`14550`**

Please update docs soon.

ilia-sheremet, 2016-01-31:
But isn't `14550` port for QGC, and `14540` for ROS? It worked for me like that before, now something has changed.

kPanesar, 2016-02-01:
I have the same issue actually. Only port 14550 works for me. Is there a way to change the port when running the simulation?

nachi1992, 2016-02-01:
I can now interface to ROS on 14540 and also connect to QGC at the same time. I just updated my px4 firmware repository to the latest one and it worked.

julianoes, 2016-02-01:
@nachi1992 Thanks for the feedback. This is now fixed with https://github.com/PX4/Firmware/commit/86c8308e984a45a689230fd611542b453a18ecd4

closed 2016-02-01 by julianoes