PX4/PX4Devguide #88: Linux Build Instructions Interpretation

petermccloud, 2016-02-15:
I know this is a nit, but in the Linux installation directions at:
http://dev.px4.io/starting-installing-linux.html

The instructions can be interpreted ambiguously. I was going through the instruction on the page beginning right at installation, before going into the hardware specific commands.
The line for installing the simulations tools, wraps around and I initially interpreted it as two separate lines. This is how it appears in the instructions:

**sudo apt-get install ant protobuf-compiler libeigen3-dev libopencv-dev openjdk-7-jdk
openjdk-7-jre clang-3.5 lldb-3.5 -y**

My mistake to interpret it that way, but formatting it similar to the command above it:
**sudo apt-get install python-argparse git-core wget zip \
    python-empy qtcreator cmake build-essential genromfs -y**
might help avoid user installation issues.

julianoes, 2016-02-15:
Hm, it is actually only one line, and just the formatting that wraps it.

julianoes, 2016-07-27:
Resolved in new gitbook.

closed 2016-07-27 by julianoes