PX4/PX4Devguide #97: Snapdragon: Clarify where to get SDK files

LorenzMeier, 2016-02-22:
@mcharleb The instructions here are very unclear when it comes to *where* to get all those files. It would be important to clarify this and make sure all the required files become available:
http://dev.px4.io/starting-installing-linux.html#snapdragon-flight

mhkabir, 2016-02-24:
It would be great if we can get this up soon. It is currently impossible for a person bringing up the board to get the toolchain up and running for the Flight, even after registering, signing license agreements and getting access to the downloads.

Especially, ```Hexagon.LLVM_linux_installer_7.2.10.bin``` seems to be unavailable from any source, and I've been blocked on this for ~3 days.

mcharleb, 2016-02-24:
@mhkabir I just made a request to have the 7.2.10 toolchain set to you

mhkabir, 2016-02-24:
@mcharleb, many thanks! :)

LorenzMeier, 2016-02-24:
Mark, is there a way to make this less one-off, or fix our docs so that step is explained better?

mcharleb, 2016-02-24:
People can contact me if they need access to the toolchain for PX4 development.

mcharleb, 2016-02-24:
The link is back up:

https://developer.qualcomm.com/software/hexagon-dsp-sdk/tool-request

Interested parties can fill in the form to request Hexagon.LLVM_linux_installer_7.2.10.bin

LorenzMeier, 2016-02-27:
I fixed the docs. Mark, you can use the edit icon in the docs to edit a page in-place without sending a PR. I would appreciate if this could be done as things change and not get pushed back. Documentation is not a separate job, but part of the development cycle. Thanks!

closed 2016-02-27 by LorenzMeier