PX4/PX4Devguide #106: Snapdragon : Camera API and docs

mhkabir, 2016-02-24:
Need to add docs on using the ```qcamvid``` application and ```camera-test``` application. This is already done in my personal notes, will clean up and send in docs soon. These applications are OK for basic testing, but not for anything like actual FPV or control. Latencies are too high, RTSP stream is non-optimal and output to stdout is broken for me.

There is also currently no way for a Linux application to get image frames directly.

Moreover, the camera API needs to become usable. As per the "Snapdragon Linux Camera Interface Specification rev A" document, the API is opened up, but no reference implementation is available openly.

FYI, @LorenzMeier @jgoppert @jywilson @mcharleb @mswingtra @devbharat 

I need to bring up the cameras on the board as soon as possible, and any help would be appreciated. I'm opening to writing the necessary drivers to interface the cameras via v4l2 and/or directly to ROS. Just the source code to the ```camera-test``` app and ```libcamera``` (both referred to in the above document) would be great to have. Impossible to implement something without them. If any of you have any information, it would be highly appreciated. I've already written to QC and Intrinsyc, but am yet to get any response.

mcharleb, 2016-02-24:
@mhkabir The Libcamera API is documented in following interface header files
 
hardware/qcom/camera/libcamera/inc/*.h
 
camera-test and camera-daemon (both are open-source) can be used as reference applications for this API. Camera-test is smallest working example of getting preview/video callbacks using libcamera API.
 
The camera-daemon source is in hardware/qcom/media repo. That contains Doxygen documentation as well. This application integrates with video encoder and live555 library to provide FPV.
https://codeaurora.org/cgit/quic/la/platform/hardware/qcom/media/tree/camera-daemon/doxy?h=LNX.LER.1.0-68046-8x74.0

Live555 is compiled as part of meta-qr-linux layer:

https://codeaurora.org/cgit/quic/lx/qr-linux/meta-qr-linux/tree/recipes-extended/live555/live555.bb?h=LNX.LER.1.0
 
Video encoder also has opensource OpenMAX API, that is part of hardware/qcom/media repo:

https://codeaurora.org/cgit/quic/la/platform/hardware/qcom/media/tree/?h=LNX.LER.1.0-68046-8x74.0

I know the above is a bit lacking in specifics but I hope it points you in the right direction.

mhkabir, 2016-02-25:
@mcharleb Thanks a lot! This should get me up and running :) 

I'd managed to reverse engineer my way to the Android libraries, but your links are gold. Thanks!

mhkabir, 2016-02-25:
@mcharleb I had a look at the repos. Looks like they are all Android implementations. I also did a search directly on CodeAurora, but couldn't find any Linux libcamera repo.

devbharat, 2016-02-25:
If plane vanilla v4l2 driver is available for the camera, then the way I have been accessing images with (for example IMX6) is through gstreamer stream piped into ROS handling both triggered-snapshot and streaming modes while also handling timesync to clock from an external controller like pixhawk. I can make the code available, all it requires is support for gstreamer in the image, and v4l2 drivers for the cameras.

mhkabir, 2016-02-25:
As far as I know, the camera has no V4L2  drivers available.

I think we have a workable solution coming up soon. Will keep this thread
updates.

mcharleb, 2016-02-25:
@mhkabir The following link is for non-android libcamera Makefile.am

https://codeaurora.org/cgit/quic/le/platform/hardware/qcom/camera/tree/libcamera/Makefile.am?h=LNX.LER.1.0

There are no fully compliant V4L2 drivers that would work with gstreamer.

Dhoeller19, 2016-03-24:
Dear all ( @mhkabir @mcharleb )
I am also working with the libcamera API. I am able to run the camera_test code on the snapdragon and I get nice images. 

I am working with the camera preview and I would like to increase the FPS to 60. When I start my program,  the code prints the correct FPS : "setting preview fps range: 60000, 60000 ( idx = 3 )"  but when I print the measured fps value pFpsAvg_ (see https://codeaurora.org/cgit/quic/le/platform/hardware/qcom/camera/tree/libcamera/test/camera_test.cpp?h=LNX.LER.1.0#n477) it systematically gives me a value around 22. I also get this low value when I set the FPS to 30. 

Is this a problem you also encounter ? 
Thank you very much,
David

ChristophTobler, 2016-04-15:
@Dhoeller19 We encounter the same. But at the moment it is not a big concern since we are using the optical flow camera. 
Hopefully they fix it...

mhkabir, 2016-04-15:
Not an issue likely. Probably you cannot do that framerate due to the exposure timings. 
Try doing manual shutter and reduce your integration time to something that  can sustain 30Hz.

Dhoeller19, 2016-04-15:
Thanks for your answers. This is also what I suspected but setting these values does not seem to be supported yet.

Dhoeller19, 2016-05-01:
I try to change the exposure time of the camera via the set command 
`params_.set("exposure-time","some_number"); ` .
 But this command does not seem to have any effect on the recordings. Is this really supported yet or could you give me some pointers on how to implement this ? @mhkabir @mcharleb 
Thank you very much,
David