PX4/PX4Devguide #109: Instructions how to build for eagle

julianoes, 2016-02-27:
We need instructions how to build the Linux image for the Snapdragon Flight, and how to flash it using fastboot.

LorenzMeier, 2016-02-27:
@mcharleb @jywilson FYI: We need good instructions in the dev guide so the platform becomes consumable.

johnfinley, 2016-03-18:
This problem seemed close enough for this issue. In https://github.com/PX4/Devguide/blob/master/book/starting-building.md, down at `make mainapp-load`, there is no makefile there. I manually did the commands in the build.ninja file and it worked; I assume there is a new way to do them via build.ninja, but I don't know what to do with that file. Similar for the `make libmainapp-load`. Otherwise everything worked, starting from absolute zero!

liamstask, 2016-03-18:
`ninja mainapp-load` and `ninja libmainapp-load` work for these, fwiw

julianoes, 2016-03-18:
@johnfinley I don't quite get what you mean. Are you saying this command does not work for you?

```
make posix_eagle_default && (cd build_posix_eagle_default && make mainapp-load) && make qurt_eagle_default && (cd build_qurt_eagle_default && make libmainapp-load)
```

johnfinley, 2016-03-18:
Correct; the "make mainapp-load" does not work because there is no makefile. "ninja mainapp-load" works.

mcharleb, 2016-03-18:
If you are building with ninja, what you suggest is likely true but not at all true if you are not using ninja.

liamstask, 2016-03-18:
`make posix_eagle_default` runs cmake with the ninja generator, so with the current instructions, ninja is indeed being used.

mcharleb, 2016-03-18:
Unless you don't have ninja installed...

liamstask, 2016-03-18:
the plot thickens!

sounds like adding a note that these commands may differ based on whether ninja is installed would be helpful.

julianoes, 2016-03-18:
Ok I'm obviously completely ignorant of ninja. So is the build system somehow using ninja if it is available. Is that a feature of cmake or of PX4?

liamstask, 2016-03-18:
px4, i believe

dagar, 2016-03-18:
It's a check in the px4 top level Makefile. If you want to try it just install ninja and delete your existing build_* directories.

julianoes, 2016-03-18:
Thanks @dagar, found it: http://dev.px4.io/starting-installing-linux-boutique.html#ninja-build-system

skye-AL, 2016-04-13:
Hi all!
A bit off-topic, but fitting this issue _Instructions how to build for eagle_:
With the current aDSP image provided by Intrinsyc, the eagle_default build fails to run on the board. A newer aDSP image needs to be used. Therefore @LorenzMeier , @julianoes , @mcharleb , is it possible to make this image somehow available such that developers don't waste time trying to make it run? I wanted to correct the devguide for that point but didn't know what to say.

julianoes, 2016-04-21:
This was done a while ago. It's now as easy as `make eagle_default upload`.

closed 2016-04-21 by julianoes